rwb27/openflexure_microscope #3: rear foot

bostjanvihar, 2016-05-18:
Hi,

I'm just working on the openflexure microscope and just wanted to let you know that the foot for ilumination part is missing from the stl files. Also, it would be great to have a list of all the parts you need in advance and how many of which...

Best wishes,
Boštjan

rwb27, 2016-05-18:
Hello,
  yes - that's on my to-do list for the next version of the instructions - the file is in the STL folder here, called rear_foot.stl, but it's missing from DocuBricks.  The DocuBricks documentation should also tell you how many of each part you need, but this seems to have worked less well than we hoped - I think I will be more explicit in the next version of the instructions.

bostjanvihar, 2016-05-18:
Thanks for the reply, the rear foot isn't in the folder here either...there are some rear foot and illumination files, but they won't fit in the main body...

rwb27, 2016-05-18:
Hi again,
  sorry - I must have mixed things up.  The newest version (for which instructions are in progress) uses a different design where the illumination and rear foot are combined.  If you're building the version on DocuBricks, you'll need the file from an older commit.  I think the link below should sort you out - if not, I did just commit it to master in the STL folder.
https://github.com/rwb27/openflexure_microscope/blob/49f4a65afaeb4797fc9b9e62185b45e27fe9d51e/stl/back_foot.stl

Do let me know if that helps.

bostjanvihar, 2016-05-18:
Hey, thanks very much =)

rwb27, 2016-06-08:
Just for the record, this is now fixed in the latest version, which is in master as of aeab81f0ac3302d8a4ee149b5794b95f786f2f72.

closed 2016-06-08 by rwb27