rwb27/openflexure_microscope #24: Instructions for different microscope versions

gpereyrairujo, 2017-04-18:
In case there are different versions of the microscope (as it was during the GOSH workshop), the documentation should have a clear reference to the different kits/versions available, so as to be sure of the parts needed for each version, and the associated assembly steps.

rwb27, 2017-04-18:
Bonus points for integrating this with a build script (#22) so we generate a BOM, DocuBricks documentation, and a folder of STL files for each possible set of options.  In the future we might have to do this on demand, from a pre-compiled library of STLs and XML fragments.

rwb27, 2019-01-21:
Since swapping to Markdown instructions, hopefully this is clearer - though still not perfect.  Once we build the instructions and add an automatic BoM, customising them should get easier.