rwb27/openflexure_microscope #25: Including warnings in the assembly instructions would be useful

gpereyrairujo, 2017-04-18:
Including a "warnings" section in the documentation could be useful. E.g., some of the 3d printed parts are very fragile and can be easily broken during the assembly. Also electronic circuits and cameras can be very fragile. Also safety warnings etc could be useful.

rwb27, 2017-04-18:
Thanks for adding these - much appreciated.  I really need to sort out the instructions, upgrade to the newest DocuBricks, etc. and I hope most of these can be fixed then, whenever "then" is...

gpereyrairujo, 2017-04-18:
You're welcome, and thank you for sharing this great project openly!

rwb27, 2017-10-03:
Update: this might become possible, as and when I hack it into DocuBricks viewer...

rwb27, 2018-12-05:
I've reworked the documentation and have one warning (for the static sensitive camera module), with plans to make "warning" boxes consistent.  I might move this issue to a repository focusing on how we generate the documentation, though.

closed 2018-12-05 by rwb27