rwb27/openflexure_microscope #26: Offline documentation

gpereyrairujo, 2017-04-18:
For many situations it is necessary to have an offline version or a printed version. The XML version downloaded from the Docubricks site does not always work, and it is not as easy to navigate (i.e. going back and forth between 'bricks') as the online website. Also, a printable pdf version (or something similar) could be useful.

rwb27, 2017-04-18:
I think this should all be possible with the new version of DocuBricks - but I've not yet been able to get hold of their revised XML spec to update the project.  I agree a PDF is a must - as is an easier-to-navigate web version.  For example, hyperlinks within the document, or "embedding" the individual sections in the top-level instructions would make life a lot easier.

rwb27, 2017-09-27:
My progress on the docubricks viewer has gone some way to making DocuBricks work offline again - it's not 100% there yet, but it means the XML can be rendered by a set of static javascript/html files.  Only issue is that, at present, I'm retrieving the XML file with JavaScript, which can't access the local hard disk - only remote files via HTTP.  If anyone can figure out a way to access the XML file so that it gets pulled into the HTML document, I can glue together the rest...

rwb27, 2017-11-21:
I've now modified the CSS for DocuBricks so that the PDF generated if you click "print" in a web browser is a reasonable solution for offline documentation.  I think that's good for now, though a way to render the XML nicely offline would still be great.  Anything further would involve either leaving docubricks or some substantial movement on an offline docubricks viewer - I think it works well enough to close the issue here for now.

closed 2017-11-21 by rwb27