rwb27/openflexure_microscope #41: Documentation improvements

davidmsachs, 2017-09-19:
The documentation could use more details about how to use the microscope with stepper motors. Just a picture of it assembled would be very helpful. Also, more specific part numbers would be useful, such as McMaster-Carr part numbers, just to make sure there's no ambiguity.

rwb27, 2017-09-19:
sure, I'll add that next time I'm working on the documentation.  In the meantime, I've added a picture of the motorised microscope here:
[image]

In addition to the parts for whatever version of the microscope you're building, you need three [small gears](https://github.com/rwb27/openflexure_microscope/blob/master/stl/small_gear.stl) ([``small_gear.scad``](https://github.com/rwb27/openflexure_microscope/blob/master/openscad/small_gear.scad)) and you should use a microscope body with motor lugs (i.e. one with -M in the name).

The motor controller pictured there is a [Fergboard](https://github.com/fr293/motor_board), but I've most recently been using a very simple [motor controller based on an Arduino Nano](https://github.com/rwb27/openflexure_nano_motor_controller).

I should revise the BoM with more details on where to get stuff, I've struggled a little to find suppliers that are available in all the continents where people want to build it, but perhaps I should just pick somewhere that has good descriptions of the parts as it's better than nothing.  I've started a wiki page on [non-printed parts](https://github.com/rwb27/openflexure_microscope/wiki/Non-printed-parts) to help with that - and I'd be delighted if you added part numbers as and when you find them.

rwb27, 2017-09-19:
PS please do ask here if there are any specific parts that are eluding you.  NB there should be a picture of the parts either in the BoM in DocuBricks, or early on in the documentation.  I know DocuBricks has some issues with the BoM (and my documentation in particular is a bit odd on that point) but I'm hoping together with the wiki page it makes sense...

rwb27, 2019-01-14:
I've now written this up in b18a4b964a680cd7e3fef1ff25290cb78f28940a, which will make it into a release very soon I hope!

closed 2019-01-14 by rwb27