rwb27/openflexure_microscope #42: Support for $3 chinese webcams

rwb27, 2017-09-19:
The cheap webcams used by e.g. [hackteria](https://hackteria.org/wiki/DIY_microscopy) would be a nice alternative for educational stuff, particularly when working in areas of the world where the £25 price tag of a Raspberry Pi camera or Logitech C270 is a problem.

rwb27, 2017-09-19:
I've tested out one of these - it works OK, and the lens was mounted in an M12 thread so I could actually use the same optics module as I use for the WaterScope chinese webcam.  I de-soldered the LEDs from the board, and had intended to use one of them for illumination - but they all seem to have stopped working...

The main issue is that, with the ~50mm tube length of the M12 optics module, the image is massively too zoomed-in.  What would be ideal is to come up with a module that places the sensor right next to the lens, which would get a zoomed-out image with similar resolution.  That would not only look better, it would also be more useful for education, where a slightly lower magnification is often desirable.

This might require making the stage a bit bigger to fit the board inside the stage rather than underneath the microscope, as it is currently.  I'm imagining a dummy optics module that finishes in a platform, the board sitting on that platform, and the lens mounted directly on top of the camera PCB.

rwb27, 2017-09-19:
@matthewlippincott do yell as and when you might be interested in this.

rwb27, 2017-10-04:
I sketched out what the mount I'm suggesting would need to do:
[image]
The lens can just be unscrewed and taped upside down to the holder on the PCB.  We'd then just need a platform that holds the camera, mounted on the dovetail.

rwb27, 2017-10-11:
I've made a start on this in the [6led_webcam](https://github.com/rwb27/openflexure_microscope/tree/6led_webcam) branch.  It's just a flat platform and needs blu tac/other mounting technology - something that clips or screws onto the camera board would be much better!

Maaphoo, 2018-01-25:
Recently I've been working on the 6 LED webcam version and a few other items. I've done the following:

- Made a Solidworks version of the microscope. I know it isn't in the open source tool chain, but it has helped me (I can't say I have an OpenSCAD brain).
- Modified the actuator columns so that they no longer require o-rings or rubber bands. Instead they have nylock nut, that keeps the thumbwheel in place. The nut in the actuator column is also more firmly anchored. 
- Added stops inside the actuator housing to limit the downward motion of the actuator columns to compensate for the fact that there are no longer feet.
- switched the illumination to something similar to what is being used by  hackteria.
- created an optics holder that fits onto a modified objective clip.
- Increased the size of the stage for the 40mm version so that the optics can fit.

**Issues I'm having:**
- There isn't much space for the modified objective clip and it can be difficult to assemble. I'm still working on getting the files adjusted so they fit together without post prossing.
- For some reason, the optics don't align vertically when they are attached. This has caused some motion in the y direction during focusing. I may have to angle the objective clip to get things to line up.

I can send STEP, STL or Solidworks files if they are of interest.

Here are some pics:

[image]
Cheek cells using the 6 LED webcam. The LEDs that come with the cam give quite a blue tint. Might be worth a few more cents to find something a bit whiter.

[image]
Just screwed onto a board with spacers in between. In Nairobi we use basically the same thing, but nicer and with a cover. 

[image]
View of how the illumination attaches. Just tapped into 2 of the led pads on the pcb. I'll use black heat shrink to cover next time instead of the blue electrical tape.

[image]
Here you can see the nylock nut and the lead screw nut in the nut trap below. Also notice that the actuator on the left is at it's max position, stopped by the limiter inside of the column.

rwb27, 2018-01-26:
Nice work @Maaphoo!  I would **love** to see your CAD files, I guess you can upload them to this issue or point me to a github repo somewhere.  Sounds like you've done a power of work and your images look fab.  STLs and Solidworks would be great, I don't have Solidworks but I'll see if I can get an educational license from somewhere.

I'm curious how well the actuators work without the extra tension from the bands; is there a "dead zone" in the middle of the stage's travel where the actuator swaps from "pushing" to "pulling", or are you able to get everything tight enough that it's not a problem?  I will look at your actuator column design with lots of interest!

Also, is there a reason you were keen to use the 40mm version of the stage - the 65mm version has longer travel and less mechanical reduction, so might be nicer to use with the slightly lower-magnification optics of the 6-LED webcam.

I have had also some issues with the optics not being perfectly vertical as well, and tracked it back to the objective clip being knocked over during printing.  The tell-tale sign was that the dovetail clip has a kink in it at the same layer where the higher set of struts attach it to the microscope frame.  I added some thin links to tie it down during printing, which helped a lot - but these may have been lost at some point.

I found the LS65 version of the microscope was big enough to fit my 6-LED webcam, though it may have been smaller than yours...

rwb27, 2018-01-26:
By the way, Nichia NSPL500DS are a nice colour of white :)  The bigger problem is controlling the white balance of the camera, which I've now got a handle on with the Raspberry Pi webcam, but might be tougher with the $3 ones, as the software drivers aren't so good...

Maaphoo, 2018-01-26:
Just set it up so that all of the files can now be be accessed at our [Grabcad Partner Space](https://workbench.grabcad.com/workbench/projects/gcdNX91ERp7bqt9L4qShbWUocFUTs5mGSUJrBRDMv18t7C#/space/gc52J7YD2yzbkF6CiBX4JVg2GCteN-_qHQ4ZVq_soT-AeQ).

Each of the main bodies is in its own folder (same design just different equations). Some of the other stuff we are using is on the partner space as well. Mainly, the box.

As for the actuators, there is a slight bit of lash when reversing direction. Maybe 15 degrees of revolution? Not too much. I tried scanning past the change from push to pull a number of times and didn't notice a dead spot. If it is there, it seems small. My guess is that if it is an issue, it would be an issue for the motorized version during automated operation. It would be much easier to evaluate the performance of this setup with a motorized microscope.

As for the stage height, I did it justto reduce the number of layers and print time. In retrospect, the ability to use the normal objective clip would be much better as the small version that I'm using doesn't have much flex and requires post processing to get everything to fit together. I'll play around with the LS65 and see if I can come up with a better solution that what I have now.

Thanks for the tip on the LEDs and the objective clip angle.

Maaphoo, 2018-01-26:
Oops forgot to mention that if you just want to view Solidworks files you can do that in Grabcad. You can also use [eDrawings](https://www.solidworks.com/sw/support/edrawings/e2_register.htm).

rwb27, 2018-01-30:
Thanks for that - it looks really nice.  15 degrees is a fair bit of backlash, so might not be an option for the motorised version, but if it makes the assembly less painful and avoids needing a bunch of parts, it might be a good trade-off for the manual ones.  I've not measured; would your optics adapter fit the LS65 stage, or would it need to be wider than that?  I think your clip is 20mm from the centre, so I'm guessing not!

Also, how do you fit the lens into the model - does it just push-fit, or do you screw it in from underneath?

Maaphoo, 2018-01-30:
Did some testing today. Looks like I had the lash issue pretty much backwards. With my home brew testing setup, I wasn't able to measure much any lash when changing directions if the axis was away from the neutral point. At the neutral point, I did record much more irregular movement.  I wonder if in addition to the switching from push to pull, strings and actuator support stubs left over from printing are hanging up on the actuator a bit. Maybe my setup was just too crude.

I hot glued a protractor to one of the axes and made a stop with a pen. Pretty home brew, but it seemed to work ok. I took a picture every 5 degrees of rotation and then measured the distance from the edge of the image to a notable item in the field of view. Here's the setup and the results:
[image]

[image]

The problem that I had with using the LS65 stage is that the webcam would have to be inserted into the microscope above the objective clip before the mounting bracket was put in place. Then the webcam would have to be attached to the bracket while "in the cage" which seemed a bit tricky. Did you have another method of assembly? Here's an image of how the LS65 looks with the 6 LED optics package in place. Definitely some interference in there:
[image]
One thought I had was to make the entire stage removable except for a frame that attaches to the legs. Then the optics could be inserted from above and the original and much better objective clip could be used.

Finally here is how the optics all fit together. Yes the objective is just press fit into place:
[image]

rwb27, 2018-05-18:
STICLab have put together an OpenSCAD version here: https://github.com/sticGit/cheap_webcam

rwb27, 2018-05-18:
Also, @Maaphoo if it's useful to you, I do have a version of the optics module that has a slightly flexible push-fit for the M12 camera lenses.  If you're having quality control issues with them not fitting, this might be of some help: https://github.com/rwb27/openflexure_microscope/blob/master/builds/optics_m12_m12_lens_LS65.stl (from optics.scad with the camera and lens set accordingly).

rwb27, 2018-05-23:
hey, that looks really nice.  I'd love to see one - any chance there might be a microscope (or even better, yourself) in Nairobi in mid-June?  I've sent an email about dates.

In terms of the push-fit, I don't have photos but I'll pop to the lab now and see if I've got the bits to take one!

rwb27, 2018-05-23:
[image]
[image]
[image]
[image]
This is for a different Chinese USB camera with M12 lens, but I'm hoping they are sufficiently similar.  NB the distance from lens to sensor is waaay too long for the $3 webcams.

Maaphoo, 2018-05-23:
Thanks for the clarification. I was having trouble seeing how it fit with the microscope I think because I was confused about the length.

Unfortunately I won't be in Nairobi, but Solomon should have a number of the new design printed by the time you arrive. We are planning on making 10 next week.

rwb27, 2019-01-21:
Some possibly-useful updates:
* I've released a new version of the microscope which fixes a bunch of problems.  It also includes a "platform" mount for the Pi camera, which is not dissimilar to your mount for the USB webcam.  We should probably adapt a version of it to mount the 6 LED camera.
* STICLab have made a version of this, there is an (old) copy of their files in [their Github repo](https://github.com/sticGit/cheap_webcam/), and a newer version on google drive somewhere (ask and I'll post it if you're interested), though that's build against a previous iteration of my stage and won't work with the current master branch...

rwb27, 2019-02-27:
As of now, the latest release can be built using a camera platform that fits the 6-LED webcam.  Currently it's just a bit of tape that holds the lens onto the webcam, I think this can be replaced in due course with a nicer adapter (e.g. STICLab's one, or Matt's one).  However, as there is now support for this on the master branch, I'm going to close this issue :)

rwb27, 2019-02-27:
The part was added in a7df420e83af84bd858f0e12f3acc1ce278ed2b3, it's possibly missing from the builds directory but will build with the build script - so it will make it into the next bugfix release.

closed 2019-02-27 by rwb27