rwb27/openflexure_microscope #43: Documentation for fluorescence microscopy

rwb27, 2017-09-25:
There are bits of code around for adding a fluorescence filter cube for epifluorescence imaging.  However, this is not documented in the assembly instructions (because it's probably too confusing).  It would be good to document the fluorescence stuff.

rwb27, 2017-09-25:
The filters I used were from [Comar Optics](http://www.comaroptics.com/components/plano-optics/beamsplitters/dichroic-beamsplitter), I used one long-pass, one short-pass, and one long-pass at 45 degrees in the [standard epifluorescence configuration](https://www.microscopyu.com/techniques/fluorescence/introduction-to-fluorescence-microscopy#figure2).  I'll add order codes here when I've found them.

The stock filter sizes are too big for the microscope; I bought larger ones and cut them down to size to fit into my [filter cube](https://github.com/rwb27/openflexure_microscope/blob/master/stl/fl_cube.stl).  If you buy the slightly larger filters you can get at least 4 filter cubes out of one set, if you're good at glass cutting (!).

JunaidBinMasood, 2017-10-10:
Dear Richard, 

Thank you for sharing the standard configurations document. I have a concern. After the Incoming Light Waves pass the excitation filter which allows the wavelengths of 450nm to 490nm (Figure 6 of standard configuration document) the Dichromatic Mirror completely reflects them onto the specimen. After reflection from specimen the wavelengths of 450nm to 490nm are not allowed to pass through the Dichromatic Mirror while rest are transmitted (as figure 6 suggests). So how is the visualization of the specimen taking place then? I mean which wavelengths will be transmitted back to the eye-piece as the 450nm to 490nm are being blocked by the Dichromatic already. Attached is the figure highlting the question in a more visual way. 

[image]

Looking forward to your reply.

JunaidBinMasood, 2017-10-10:
If you could please suggest where the Dichromatic Mirror was ordered from? The filters were ordered from Comar I saw.Waiting for your order codes. Thank you.

rwb27, 2017-10-10:
Hello, in response to your question on the diagram above, the principle of fluorescence microscopy is that you absolutely don't want any of the excitation (illumination) light to come back to your camera port.  The light that you are using to see the sample is the fluorescence emitted from the sample, which will be at longer wavelengths than the excitation light - in the diagram you pasted in, the spectrum of the sample's fluorescence is the white curve, which is predominantly green light.  This should pass straight through the beamsplitter and the emission filter.

rwb27, 2017-10-10:
The filters were all from Comar - the beamsplitter is a [long pass filter at 45 degrees](http://www.comaroptics.com/components/plano-optics/beamsplitters/dichroic-beamsplitter/long-pass-dichroic-filters), the emission and excitation filters are I think both [bandpass filters used at normal incidence](http://www.comaroptics.com/components/plano-optics/beamsplitters/dichroic-beamsplitter/band-pass-dichroic-filters).  Exactly which ones you need will depend on the dye you are using and the wavelength of your excitation LED.

JunaidBinMasood, 2017-10-10:
I re-consulted and found that it is the Strokes' shift which causes the reflected beam from specimen to go to a higher wavelength. Thank you for replying.

kevincvpp, 2018-01-03:
Nowadays, I am working on a project of a low cost lensfree imaging and the fluorescence is a big challenge. I would like to share this configuration using a prism such as it has been proposed by Ozcan (Lab Chip. 2010 April 7, Wide field-of-view lens-free fluorescent imaging on a chip) for build a lensfree fluorescence imaging. Usually, in lensfree imaging, it is not practical to use filter cubes. What do you think?

kevincvpp, 2018-01-03:
https://www.ncbi.nlm.nih.gov/pmc/articles/PMC2863091/pdf/nihms194845.pdf

rwb27, 2018-01-03:
That's a really elegant way to do it - I'd not come across that before.  If photobleaching isn't a problem, it should get pretty good rejection of the excitation light without needing expensive filters, as indeed they say in the article.  That said, if you've already got a field of view of 25x35mm, you probably don't need a mechanical stage like this one...

The filter cube in this microscope does the same job as the prism in the article you reference - essentially filtering out the excitation light.  Using the cube, excitation light passes through the sample and out the back, so it doesn't hit the sensor (i.e. you excite and detect from the *same* side of the sample).  This is roughly equivalent to the prism arrangement in the paper; it also stops excitation light from passing through and flooding the sensor.  The difference is, of course, that the prism allows the sensor to be much closer to the sample, which you need for lens free imaging.

In a lot of applications, illuminating and detecting through the same objective is convenient and helpful, because it means the intensity in the sample is quite high (because the objective lens concentrates the excitation light very strongly).  That's why the openflexure microscope uses a cube.  If your application works nicely with a lens free imager, the prism method may well be better - I guess it should generate quite nice dark-field images too, if the filter is removed.

davidmsachs, 2018-05-14:
Thanks, that's very helpful! 

I also want to confirm some stuff about the specific LEDs and filters. I didn't see any specifics in this repository, but I found some documentation from that 2015 iGem project that I'm assuming you were somewhat involved in. They have this in their instructions:

1 Blue LED (GFP) Mfr: Kingbright; Part No. L-10934VBC/DS-D 
1 550nm dichroic mirror, 25x16mm (GFP) Mfr: Comar; Part No. 550 1Y 116 
1 490nm excitation filter, 25x16mm (GFP) Mfr: Comar; Part No. 495 1K 116 
1 500nm emission filter, 25x16mm (GFP) Mfr: Comar; Part No. 515 1B 116

Is that what you'd recommend for GFP? For RFP, I noticed some attempts on their website, followed by: "Note: this did not work in practice". Have you had any success with RFP? And if so, do you have recommendations about specific filters and LEDs?

Incidentally, I've been printing stuff with black PLA/PHA from Meltink, which has a matte finish. I find that it helps a little for reducing any unwanted reflection.

Thanks again,
David

rwb27, 2018-05-15:
No worries.  Those look like the right filters - but they aren't the right size (I designed it to be quite small to fit with the rest of the optics).  I usually ordered them deliberately too big, then used a cutting wheel/diamond scribe to scratch a line across the glass and snap it to size.  12x16mm is about right, but check by measuring the holder first!

I've only ever tried with GFP/Fluorescein wavelengths, so can't comment on RFP, other than saying that Comar have a pretty wide range of dichroics and you can find LEDs at most useful wavelengths.

rachelaronoff, 2019-03-13:
I want to acquire tiled fluorescence images of 'comet cell' assays, so don't need huge magnifications, and we use SYBR safe for green fluorescence.  I have an NPL FLUOTAR 10x objective available (0.3) from Leitz, and wonder for the optics parts, if that will be sufficient. However, is it the piece f40d16 or the f50d13 that would be the most appropriate to make?  It is clear that one is shorter than the other, of course...  (next step before printing - to see if our black filament is matt or shiny!) . Thanks!

rwb27, 2019-03-13:
Hi @rachelaronoff, you do need one extra lens to make it work properly; the design is meant to use a "finite conjugates" microscope objective, together with a secondary lens to shorten the distance to the sensor and adjust the image size to fit the Pi camera.  The two pre-compiled options are:
* a 12.7mm diameter, 50mm focal length achromatic lens, e.g. ThorLabs AC127-050-A - fits ``f50d13``
* a 16mm diameter, 40mm focal length lens, e.g. Comar 40 PC 16 (I think) - fits ``f40d16``
I reccommend the ThorLabs lens, it is slightly more expensive, but gives better quality images.

If you have an "infinity corrected" lens, you are lucky (these are generally more expensive and better quality) but it won't work properly with either of the options; you will still need one of the above lenses, but would need to compile a custom optics module.  If that's the case, let me know and I'll compile one for you - it would be no bad thing to add this to the build script.

If you are not sure whether your objective is infinity corrected or finite conjugates, look at the writing on the side of it.  If the number 160 appears anywhere, it is probably a standard finite-conjugates objective.  If the infinity symbol is present, it's probably infinity corrected.  Most RMS-threaded objectives (i.e. ones that will fit the plastic part) are finite conjugates.  Modern infinity-corrected lenses tend to have larger threads.

Lastly, you can build the microscope without the additional lens.  It will still produce images, but they will be of lower quality and the magnification will be different.  I don't reccommend this, but it's possible it will work for your application.

rachelaronoff, 2019-03-13:
ok, I will order the Thor labs one, and make the f50d13 piece.  
There is no infinity sign on the objective - it says 170 / -, NPL FLUOTAR 10/0.30 (the last is the numerical aperture, I believe)...

MakerTobey, 2019-03-14:
I FINALLY found this issue - Yeahh :-) I have been looking around for parts, light sources and affordable filters quite a lot recently and came to very similar conclusions as you above, for example that Comar has great material to offer. Just the sizes of the filters I bought might be a problem now.

I described the (here relevant) part of our project [here on GitHub](https://github.com/MakerTobey/OpenMicrofluidics/tree/master/Open%20FADS). There may be other interesting information to you. In our project we develop and open hardware platform to sort microfluidic droplets based on their fluorescence.

The question about what light source to choose came up several times. There are a lot of options but soldering surface mount LEDs, finding suitable heatsinks and lenses etc. can be a real issue! Also what wavelength to start with? 488nm is perhaps the most important standard laser-based excitation wavelength and lots of dyes work with it. Here I share our LED based solution to work with that wavelength:

We purchased the folloing from Luxeonstar.com (February 2019):

Part Nr | Description | Price
--- | --- | ---
SP-01-B6 | LED: Blue (470nm) Rebel LED (LXML-PB02) on a SinkPAD-II 20mm Star Base - 74lm @ 700mA  | 13.08 $US
10193 | Carclo 8.7° 20 mm Circular Beam Optic - No Holder | 2.60 $US
10235 | Carclo 20 mm Black Round Optic Holder - Flat Bottom | 0.70 $US
LXT-S-12 | Pre-Cut, Thermal Adhesive Tape for 20 mm Star LED Assemblies - (12 Piece Sheet)| 7.49 $US
N30-10B | 30 mm Square x 10 mm High Alpha Heat Sink - 14.0 °C/W 10193 Carclo 8.7° 20 mm Circular Beam Optic - No Holder | 5.72 $US
LP-01 | Assembly Press for Mounting Single Rebel LED Assemblies to a Heatsink | 0.00 $US
shipping | Shipping from Canada to Europe | 21.01 $USD

This LED is a nice all-round solution and the wavelength (470nm) is close to 488nm - the filters further cut out the desired part of the spectrum.

I also choose Comar filters that allow fluorescence detection much closer to the excitation wavelength. This makes it first of all much more efficient to detect signals from few molecules, and secondly I intend to use some dyes such as the bacteria stain SytoBC that emits mainly around 498nm, so very close to the excitation of 488nm.

MakerTobey, 2019-03-14:
> If you are not sure whether your objective is infinity corrected or finite conjugates, look at the writing on the side of it. If the number 160 appears anywhere, it is probably a standard [...]

I would be interested in such an adapter for an infinity corrected objective :-) . Is there actually still any benefit in using such a more-expensive objective here? Meaning, is there still a quality benefit if combined with another lens?

rwb27, 2019-03-14:
nice, thanks for posting :)  I've used the Luxeon Rebel series LEDs in the past and really liked them - I have mostly just used the simple 5mm plastic cased ones on this microscope, but something like that would be a big improvement once I add some proper condenser optics for the fluorescence path.

In fact, I have just been working on a better reflected-light illuminator (in this case for imaging graphene) that would also work for fluorescence - I am uploading as I go but there's not any documentation to accompany the STLs yet...

rwb27, 2019-03-14:
> I would be interested in such an adapter for an infinity corrected objective :-) . Is there actually still any benefit in using such a more-expensive objective here? Meaning, is there still a [...]

Might be one to add to a new issue, or #112.  It's not a big job to compile as a one-off (there are in fact comments in optics.scad explaining it), but it would take a few lines of code to do it more neatly.  For sure the infinity corrected lenses are likely to be better quality, and also to make it possible to do some other fun stuff.  For most folk the cost outweighs the benefits - but for soem applications it will be worthwhile.

rwb27, 2019-03-14:
The illuminator is on the better_reflection_illuminator branch btw

MakerTobey, 2019-03-14:
> The illuminator is on the better_reflection_illuminator branch btw

Found it.
[image]

But this this for an upright microscope? Or for using an objective on top (=condenser) as well as below the sample (for imaging)?

P.s.: Do you have a tip where to get high-quality budget objectives from? I found [this](https://www.microscope.com/accessories/objective-lenses) but there are is probably better ones.

rachelaronoff, 2019-03-14:
thx for all this discussion that will help me greatly in this build! (hi, Tobey!)
but what about the tip to use matt black for the optics printing? (ours looks super shiny to me) which filament do you use?
thx!

***
Rachel Aronoff, PhD
Directrice Scientifique 
AGiR!  (www.genomicintegrity.org)
Action pour l'intégrité Génomique via la Recherche! 
&
Présidente et
Responsable Biosécurité
Hackuarium (www.hackuarium.ch)
wiki.hackuarium.ch
#DIYbio #citizenscience #DoItTogether

rwb27, 2019-03-14:
Hmm, perhaps I should remove the "matt" - I just buy black stuff from RS components, I've not found any particuarly non-shiny filament.  It's not too big an issue in the regular microscope, but for some of the reflective microscopy I'm working on now it's a big pain.  My solution so far has been to add some black fabric to the places where scattered light is a big problem.

MakerTobey, 2019-03-15:
I would like to follow up on my ideas more, to clarify and perhaps inspire different options, because I think what I am trying to achieve is slightly different from other builds because I need to integrate an extra sensor. I hope this is not making it more difficult for other users to follow the thread.

I need to use a filter cube not to couple illumination and (fluorescence) imaging, but I need the two light paths either for 
(a) (normal) imaging + fluorescence sensor (normally a photomultiplier (PMT), but we are working on an open source avalanche photodiode (APD) module instead) or
(b) two different fluorescence sensors and no imaging.
That is also why I do not use a dichroic mirror, but a beamsplitter for my cube. (See in respective [GitHub repro](
https://github.com/MakerTobey/OpenMicrofluidics/tree/master/Open%20FADS) on Open Microfluidics))

This means for the illumination it may be better for me to either use a top-illumination (similar to reflection microscopy but with condenser) or couple the LED light right into my PDMS chip that I am imaging (by direct contact from the side, or a laser beam via a PDMS prism on top of the substrate). I assume it could be a problem that my 'specimen' are droplets on the surface of a glass slide, but there is also an un-reproducibly thick layer (ca. 1cm) of PDMS on top, which may make a focussed illumination difficult.

Alternatively I could imagine a more complicated set up, where white imaging light comes from the top, and an extra fluorescence illumination from the bottom side. This may be better in terms of usage, alignment and focus of the fluorescent light source, but would require most Open Felxturescope redesign.

rwb27, 2019-03-15:
Hi @MakerTobey it might indeed be helpful to move this to a new issue... could you start one?

AlEdwardsUoR, 2019-03-19:
Regarding 'non-shiny' filament:

We've had great success printing matt (or is it matte) finish materials to make dark boxes for various imaging projects, by accident.

We’ve found a ‘PLA pro' filament from Ooznest in the UK prints with a matt finish if you use ‘standard’ PLA settings i.e. hotend temp. If you increase hotend print temperature, eventually it becomes shiny. This makes sense if you think about polymer flowability vs temperature.
 
No idea how much it will help reduce reflection. 
 
I’m fairly sure it’s printing matt because it has a higher melting temp and is supposed to be printed at higher than 'normal' PLA. We were just too lazy to fiddle the hotend temperature when we first bought it, and used generic PLA print settings. It seems to have good mechanical strength and layer adhesion even though we’re printing below recommended temp. 
 
It's likely that other PLA filament might also give matt finish with reduced temp, although it might have other undesirable consequences.

What I can't tell you is relative reflectiveness of the different printed parts, but whenever we are making a 'dark box' for imaging we use this PLA.

The other trick for making things matt and black, which we've also used with open flexure optics, is to simply spray paint with matt paint- acrylic matt paint from Ambersil works great. It's smelly and messy but blocks light effectively, so I printed cream PLA lens tubes and spray painted inside to block the 'glow'.

rachelaronoff, 2019-03-26:
Just a quick update and concern: we have made most parts successfully (and they look really super!), but the filter cube gave me some trouble - after my third try (and adding supports) our Prusa maybe made an ok job of it.  But now the filter cube holder also messed things up. (or maybe the nozzle just got clogged)  After the easy experience with everything else, I wonder if I got the correct stl files.  Hope to get the chance for a second try with the cube holder before the end of the week... Finally, another thing to note - the current filament is called 'galaxy' with little white specks!  (hopefully non-reflective!)

rwb27, 2019-03-27:
hmm, the filter cube isn't quite as carefully optimised as the rest of the microscope, it's true - might be worth opening a new issue about that with some photos, I'm curious how it went wrong.  I'm also not quite sure what you mean by the filter cube holder - is that the optics module? or do you mean the LED holder?

rachelaronoff, 2019-03-28:
hello!
the files I printed were called fl_cube and fl_cube_outer, but the 'outer' piece is smaller than the cube.
maybe I did something wrong when I made the gcode (esp with the supportive extra?)?  and am thinking about trying again.  I should also note that we changed the nozzle on the printer, and it could be that it is better now...

I will attach a pic with 2 of the cubes and one 'outer' piece...
[image]

rachelaronoff, 2019-03-28:
again, these are done in the galaxy filament which has sparkles!  (another reason to try again ;)

rwb27, 2019-03-28:
Hi @rachelaronoff, the fl_cube_outer is an old test piece and you can safely ignore it.  The two parts you want are the filter cube, and the LED holder.  I think I printed my one rotated 90 degrees from that position, which then means you don't need any support material.  The face of the cube that doesn't have a hole in it should be on the bottom, IIRC.

rachelaronoff, 2019-03-29:
yes, we had made a flipped cube too, to avoid having the hole at the bottom, but someone else told me to try it with the supportive material...  
Will go for it and not worry.
(looks like I will have a matt green filament now...)
The file of the LED holder, is it 'illumination_and_rear_foot...stl' ?

rwb27, 2019-03-29:
It is https://github.com/rwb27/openflexure_microscope/blob/master/stl/led_holder_5mm_shorter.stl

Sorry that’s not very obvious!

rachelaronoff, 2019-03-29:
Thanks!
but, do I still also need an 'illumination and rear foot' piece?  (I already put it flat for the gcode).
making the led holder next as you recommend... (btw, the cube turned out very well, printed on its back, as you recommended)...

rwb27, 2019-03-29:
Not if you're using the latest release - that part is obsolete (assuming you're using the microscope stand).  If you're not using the microscope stand, you can print the rear foot from the builds folder.

rachelaronoff, 2019-03-29:
have the stand, so no need!
thx again!

rachelaronoff, 2019-04-09:
this is to remind you to fix the condenser lens specs - from about 12mm in the current doc to ->  13mm diameter, 5mm focal length plastic lens.
thx!

rwb27, 2019-04-30:
The repository is [moving to GitLab](https://gitlab.com/openflexure/openflexure-microscope/), and this issue has been migrated.  I'll close all the issues here in due course, but am leaving notices on all the currently-open ones.  If you head over to the other repository, this issue will be updated there.

rachelaronoff, 2019-04-30:
thank you for this info!  Is being in GitLab a big step up? ;)
have you put together something more about the fluorescence build?
(can one readily image in bright field and then epifluor on the same rig,
or not??)
I wrote to Valerian about the sangaboard, and hope he will respond soon,
but Kaspar said this may not be so quick...
:)

rwb27, 2019-04-30:
GitLab works much the same - a few improved features, but mostly it's just so that we have everything in one place.

I thought I answered your question on swapping between BF and fluorescence somewhere else - it is possible; if you leave the beamsplitter/second LED in place, then turning on the (usually blue) LED will give you fluorescence, and turning it off and turning on the LED at the top will give you transmission.  Of course, as the emission filter/dichroic are still in the beam path, you will get a brightfield image in the colour of your fluorescence emission.

Valerian isn't always super quick on email, I'll send him a whatsapp message to ask him to look out for you though.

R

rachelaronoff, 2019-04-30:
super!
that is just what I thought, but I didn't remember getting a direct
confirmation. (I think the question came while you were still at the
African OSH...)
Also, btw - Kaspar mentioned that the v3 is harder to put together and that
Valerian might be able to have it already assembled.
next I get to test whether my matt black spray paint (somehow the Prusa had
red in, when I was ready to make the right optics part ;) still allows the
objective to readily screw in! :)
best,
Rachel

rwb27, 2019-04-30:
great.  Valerian has some assembled boards (they are all-surface-mount, so he got them built by Seeed) that I'm sure he'd love to sell.  I messaged him and he hadn't seen your email, hopefully he'll appear on this thread.