rwb27/openflexure_microscope #44: Tall feet + Base plate

GraceAnyelwisye, 2017-09-27:
When am using tall feet the base plate does not work.

rwb27, 2017-09-27:
hmm - I think the problem is that, with the taller feet, the bottom of the cups for the feet is below the bottom of the baseplate.  The lines of code that control this are at the top of the SCAD file:
`
mount_z = 30; //height for the mounting surfaces
foot_z = mount_z - foot_height; //height of the bottom of the feet
`
Try increasing `mount_z` - that should probably fix it.  What might be better is to define them the other way around, so that you set `foot_z = wall_h`, then `mount_z = foot_z + foot_height`.  That way, the baseplate is always tall enough, no matter what foot_height is set to.

Why don't you give that a go, then maybe make a pull request :)

rwb27, 2017-09-27:
Fixed with pull request #45.

closed 2017-09-27 by rwb27