rwb27/openflexure_microscope #46: 3D Printing - too much supportive material required

wcolemanred, 2017-10-23:
Two different 3D printing companies have reported that tube_lens_tool, optics_picam2_rms and body_ss have thin walls while optics_picam2 and body_ss  and have overhands, all which will require lots of support structure. 

One 3D printer has suggested that 45 degree supports on any overhands are added to enable printing without support structures.

Is that possible?

rwb27, 2017-10-23:
Hmm - it would be fine to make a thicker version of the tube lens tool if that is a problem (it's just a cylinder slightly smaller than the lens, to allow you to push it into the holder).  Looking at ``openscad/tube_lens_tool.scad``, the wall is 1mm thick which ought to be OK for most printers - did they give you a figure for the minimum wall thickness?

I print both of the parts you mention that have overhangs on my printers (an Ultimaker 2+ and a RepRapPro Ormerod) with no support material, and no problems.  They have been designed very carefully to print OK despite the bridges (there aren't any "overhangs", only "bridges" - this is an important distinction, because bridges (i.e. where you are printing between two walls) are generally possible whereas overhangs (e.g. printing a diving board) generally aren't.

The microscope design has been through a lot of iterations to get it to print nicely without support.  If you just analyse it with an automated tool, it will usually tell you that support is required.  However, it's not, at least not using a RepRap-type plastic extrusion 3D printer.

Do let me know how you get on - I've never actually had them printed by an external service, so it would be interesting to know how it turns out and (if they don't mind you sharing it) how much it costs.

rwb27, 2017-11-13:
Hello, I was just wondering if you'd gotten any further with this one?  It would be good to know if you managed to get one printed in the end...

rwb27, 2017-11-21:
I'm closing the issue as I've not heard back - but do re-open it if you have anything further to add.

closed 2017-11-21 by rwb27