rwb27/openflexure_microscope #50: More details on printing settings please.

sctv, 2017-12-14:
Hi, this is an amazing project.
I am wondering if their are there any more detailed printing instructions than:

> I usually print with a layer size of 0.24mm on my Ormerod, which takes 8 hours for the main structure. “low” quality on an Ultimaker 2 (0.15mm layers) produced similar results in about 5 hours. If your printer has a standard-sized bed (180mmx180mm should be fine) then it should be possible to print the complete microscope in one go. [from openflexure_microscope_5_16_beta2\docs\openflexure microscope 2.docubricks.xml]

I am specifically thinking about fill which might have a big effect on the mechanics/strength/flexibility of the resulting print.   

Da Vinci 1.0 1.75 mm, ABS,  octoprint  
Printing just_leg_test.stl with: 
Print temperature:     230°C
Print bed temperature: 90°C
Layer height: 0.2 mm
Fill density: 20 %
Shell thickness: 1.2  mm
Print speed: 60 mm/s
Top/bottom: 0.6 mm
No Support

Took 33 minutes and 1.97 cc plastic
The overhangs weren't perfect, but the leg flexes ok. 

Still, before I embark on a 4-5hr print, it would be good to have a few more details. 

Thanks. 

Thanks

rwb27, 2017-12-14:
The guidance on printing is a bit scattered, sorry!  I usually print with 33% infill, but the microscope is mostly perimeter anyway, so the infill doesn't affect it very much.  I think your print settings should be fine - a little sag under the bridges shouldn't stop it printing nicely, so I'd say go for it!

I thought I had put better instructions in the documentation than that, but looking at it again I haven't mentioned infill and it would be helpful to do so.  I'll update that.

Do let me know how you get on :)  Print/build photos are definitely encouraged!

Richard

sctv, 2017-12-14:
The test leg print at 25% fill.

rwb27, 2017-12-14:
sorry - should there have been an attachment?

sctv, 2017-12-14:
Was supposed to be....will add a photo some other way.

rwb27, 2017-12-15:
Were you replying to an email? If so, I think GitHub loses attachments. If you click the link to view on the Web, you should be able to paste in /upload that way.

sctv, 2017-12-15:
That's what happened.

rwb27, 2018-01-30:
I think we've resolved this one... Do re-open if you have more questions.

closed 2018-01-30 by rwb27

rwb27, 2018-01-31:
super.  I do need to document how to check it's printed OK a bit better - in fact that was what I spent part of yesterday doing, resulting in a couple of hopefully-useful reference sheets on [the names of bits of the microscope](https://github.com/rwb27/openflexure_microscope/blob/master/docs/images/microscope_body_part_names.pdf) and the [quality checks](https://github.com/rwb27/openflexure_microscope/blob/master/docs/images/microscope_body_quality_checks.pdf) I do when I print one.  I should probably indicate how far the leg should be able to bend - perhaps I could integrate a little end-stop, that might be nice.