rwb27/openflexure_microscope #52: Illumination Arm Question

SuAdams, 2017-12-22:
Working on the WEF project, and have concerns about whether we have the correct illumination arm for the body.

Both were the SS models, but the illumination arm is far higher than depicted in any of the photographs which you have.  I wanted to check that the SS arm and SS body which we had printed from the 5.16 zip file and are correct. 

Also, I'm not sure whether the focus is on the sample itself; I'm concerned that it is on sample slide/glass instead.  Do you have any advice on this?

FYI, I have built 2 versions of the microscope which we've had independantly printed.  Of the two versions, one version is complete, the other, just the main body and actuators.  This is because one is low quality and the other high quality.  You may be interested to know that whilst the low quaity version works (with just 2 parts needing the high quality) the high quality print is far easier to assemble.

We also found it very difficult to get hold of hex head screws which were also hex socket.  I went for a 30mm hex socket cap head solution with an additional nut at the top of the screw to resolve.

I've made notes of the build and would be happy to share.

rwb27, 2017-12-22:
I think that might be my fault - sorry!  As there are now over 100 possible configurations of the microscope, it gets increasingly hard to make sure the right bits are compiled with the right filenames, so I've switched to an automatic build system.  I've not yet built that into "release" archives, but if you head to the [builds folder](https://github.com/rwb27/openflexure_microscope/tree/master/builds) and grab the SS version of the illumination from there, it should be correct.

I do need to update the release version, but I keep putting it off because I think I'll have the build server running soon!

If you're using a Raspberry Pi camera module v2, I've found that you can't focus all the way through a standard-thickness microscope slide.  That means you usually need to either put the sample on the bottom of the slide (which is easier if you use a coverglass) or use a thinner slide/large coverslip.  The first version of the camera module had lower resolution, but a longer working distance, so you could focus through a slide (which was very useful).

I'd be curious to know what print settings/nozzle size correspond to "low" and "high" quality, and which bits you needed to print high quality.

Getting hold of hex head screws that small is annoying, I know - but your  solution (which I'm hoping you saw in the documentation) works just as well.  I buy mine from Anglian Fasteners, but you need to order quite a few to make it worth the postage!  I tend to buy a few hundred at a time...

Please do send me your notes - or even better upload them as a [build report issue](https://github.com/rwb27/openflexure_microscope/issues/new?labels=build%20report).

rwb27, 2018-01-09:
I don't think there's anything outstanding here, so I'll mark this one as "closed" for now.

closed 2018-01-09 by rwb27