rwb27/openflexure_microscope #61: Robo R1 plus - printing issue

TanOneDeg, 2018-01-11:
I've had this printer for about 1 1/2 years now with no printing issues and have printed quite a few things of my own design.  Since I've had no problems with the printer, there never was a need to manipulate any of the settings other than the heat. For starters I don't know how any 3D printer of 'this type' would be able to print the horizontal members between the columns on the test print since it can't extrude anything onto air. Even so I gave it a try and as I expected it didn't print.  I'm still using the slicer that shipped with the Robo R1 plus which is MatterControl. I realize better slicers exist. My Robo has printed everything I've designed, including molds, gears etc., with no issues. Then again everything I've designed didn't require supports of any type. 

So I printed the test .STL and have attached an image of it. I didn't scale it, just printed it. Even if I would have printed a half dozen of these at once to give each one more time to cool, I still don't see how the horizontal members would ever print without some sort of support under them.  

Since one of my hobbies is micro/macro photography (mostly imaging diatoms with a microscope) I thought the openflexure microscope project would be interesting to try (as time allows).

 https://www.matterhackers.com/store/l/mattercontrol/sk/MKZGTDW6

Regards ...

[image]

rwb27, 2018-01-11:
I think we have different ideas of "didn't print" so thanks for the photo!  While the printed part isn't a perfect representation of the STL file, I would say that your test piece **did** print OK.  When bridging over an air gap, there will always be a little sag/spaghetti underneath (though it can be reduced if you keep the temperature relatively low, don't print too fast, and use a slicer that handles bridges well).  The top bridge (which is the only one that matters) looks perfectly solid, although I've seen neater.

The microscope has been designed to tolerate printing artefacts, like rough surface quality and gunk under bridges - so I would be hopeful that it will still work nicely if you print it on your machine with whatever settings you used for the test piece.  If you wanted to try for higher quality, I'd suggest giving it a go with slightly lower temperature - that helps the plastic to stretch over the bridge without sagging or forming loops.

The two thin bridges are only there to stop the legs being knocked over during printing and don't serve any purpose afterwards, so if they look ugly you can snap them out with pliers quite easily.  The bridge at the top is important, but while it's not the neatest bridging I've seen, it has also spanned the gap pretty well.

[image]
I've attached a photo of a part I had on my desk to show what the part looks like on my Ultimaker 2+.  You can see there is still a little sagging and a few loops of plastic under the main stage, but the bridges have printed without too much junk underneath them.

TanOneDeg, 2018-01-14:
The top bridge (not the block on top of it) didn't print very well in my opinion. May I ask why the top bridge, or even the others for that matter, are horizontal instead of arched (gusseted support)? Seem like the top bridge would print perfectly and even be stronger/stiffer if done so.  Naturally I will give it another shot. I did drop the temperature down to 191c on my first attempt (using PLA).

rwb27, 2018-01-16:
I think it wasn't brilliant, but it was still good enough to work :)

There is a very good reason to keep the bridge horizontal, which is that it's intentionally not rigid; the motion of the stage comes from bending the thin bits of plastic at either end of the horizontal part (you should be able to distort the test leg print into a parallelogram).  Using an arch shaped bridge would make it impossible to have those very thin spans either side of the thick central strut.  If you can distort the structure so the angles are about 6 degrees away from right angles, without the thin bits breaking, that's the most important test.

Also, bear in mind that arched supports won't print nicely after about 45 degrees, so it's often preferable to continue up at 45 degrees then bridge across the top, even when the flexure hinges aren't needed.

If you're already printing at 191C I retract my comment that you might want to print cooler!  It's possible you'll find the whole thing just works better if you print the microscope body rather than the test object - the trouble with small, quick prints is that the time per layer is very short, so the object doesn't have time to set fully before printing the next layer.  I've often seen small parts print quite poorly, then printed multiple copies of the same thing and had it come out much, much better.

TanOneDeg, 2018-01-19:
Touching base with you Richard, 

Just to let you know, I went to the Robo R1 plus forum for suggestions on printing this.  A couple there that have extensive printing experience are going to give it a shot. None of them use the slicer that I do with the majority are using Simplify3D.  As I posted there, I'm not trying to beat a dead horse to death. I printed the **main_body_LS65.stl** last night, acquired from the Github site, with no supports, temp set at 189c using MatterControl and this is how it turned out.  Obviously it isn't cleaned up.  This is how it looks right off the print bed. I didn't take the time to use the proper diffusion to eliminate the specular highlights in the photographs below. 

[image]
[image]
[image]
[image]

 Of course I'm not happy with the results but I am surprised the entire model printed.

rwb27, 2018-01-20:
Thanks for posting the pictures - that's really helpful.  It looks like a perfectly good print to me, obviously the thin stringy bits aren't beautiful, but they should be easy enough to clean out and won't affect the function of the microscope stage.  Normally when I've had stringy bits like that it's because I'm printing too hot (but at 189C I'd be surprised if that's the case for you) or not using enough retraction.  However they are usually just a cosmetic defect.  I'd consider your print a solid success :)  Do post if you build it into a working microscope, and if you have a link to the R1 forum when anyone posts their results I'd also be keen to see it :)

All the best,

Richard

TanOneDeg, 2018-01-20:
I will let you know the results of others that may print this.  I wasn't happy with the model at all. The stringy stuff will be super easy to remove and doesn't have an effect on the final print. There are other areas that are unacceptable (to me)  and difficult to photograph.  With respect to your design, I still think designing gussets (probably around 30 degrees or so) at key places would be helpful in printing, especially on the top horizontal members on the towers and maybe incorporated in the supports for the physical stage platform.  Is the 3D model available for editing and downloading?  I will still experiment with my printer, I haven't given up. 

https://www.matterhackers.com/news/how-to-use-support-material-part-1

rwb27, 2018-01-22:
One of your comments seems to have vanished - that's odd!  The CAD files are all in this repository, in the OpenSCAD folder.  They are made in OpenSCAD, a free script-based CAD program.  You will need to download all the files and put them in the same folder, as the main body depends on several other files.  If you can find ways to make the print neater/more reliable by adding supports (whether they can flex with the structure or need to be snapped off after printing) I would be very interested.  It prints pretty well now but there is always room to do better.

The missing post also contained a link to a really helpful page on what printers can and can't do, including a discussion of bridging and support in various links:
https://www.matterhackers.com/news/how-to-use-support-material-part-1

TanOneDeg, 2018-03-06:
Richard,
Do you have pictures and/or a video of a working Openflexure? Just wondering. 
Regards,
-JW:

TanOneDeg, 2018-03-12:
Could this be printed with a SLA printer and work as designed?

rwb27, 2018-03-12:
I can't see any reason why not - though I don't have one to test with.  I guess you'd need to make sure the resin is sufficiently flexible, but I would guess that even the relatively rigid resins should be OK.  If you have access to an SLA printer and try it out, let me know :)

TanOneDeg, 2018-03-12:
... I was thinking Standard Nylon (SLS) PA 12 with 50um layer height?

TanOneDeg, 2018-03-12:
Never mind ... looks like they can't print it without supports and said they would be hard to remove.

rwb27, 2018-03-13:
Hmm, that's a shame.  It was put together very much with fused filament in mind, so that's not a huge surprise.  That said, the supports usually used for SLA are much easier to use than the ones for fused filament, so they might not be such a problem.  I've been told in the past that the large flat base of the body (which is necessary for adhesion on a filament deposition printer) could be a problem with SLA; I'm not sure if there's an easy way round that - though possibly printing it in a different orientation might help.

TanOneDeg, 2018-03-18:
Richard,

I cleaned up the flexure stage today but I haven't a clue if everything printed correctly. I did remove all of those braces since they didn't print to begin with.  Everything seems to flex properly when I press the flexure tabs on the bottom of the stage.  After looking at the image below and if you agree that it appears to be printed correctly what specific items would you suggest I print next? Also is there anything I should check for printing failures on the stage below?

[image]

Thanks again,
-JW:

rwb27, 2018-03-19:
Hello - that looks like an absolutely beautiful part to me!  I put together a little list of [quality checks](https://github.com/rwb27/openflexure_microscope/blob/master/docs/images/microscope_body_quality_checks.pdf) for some folk working on establishing a baseline of prints to measure reliability, etc. (I really ought to put this somewhere more obvious), I think yours will pass all the checks with flying colours.

Next, you'll need feet, gears, an illumination arm, and an optics module.  The exact files you need will depend on what camera and lens you are planning to use - I have tried to detail all this in the readme in the [builds folder](https://github.com/rwb27/openflexure_microscope/tree/master/builds/) but feel free to ask if you're stuck.  One of these days I will get round to making a form that simplifies the process of choosing the parts...

closed 2018-12-05 by rwb27