rwb27/openflexure_microscope #68: Question about high resolution optics module setup

zachjacobs, 2018-02-14:
Hello,

Great work here!  I'm still a bit confused about how the optics will work with a given objective lens.  I have found [objective lenses](https://www.amazon.com/dp/B01G5W0UH4/ref=psdc_3117833011_t1_B00FG8BC62) that are RMS compatible, and designed for 160mm tube-length finite-conjugate microscopes.  I am planning on pairing these with this [50mm tube lens](https://www.thorlabs.com/thorproduct.cfm?partnumber=AC127-050-A) from thorlabs.  This leads me to believe that I should be using the optics_picamera_2_rms_f50d13_LS65.stl file to put this assembly on a raspberry pi 2 camera.  Is that correct?

When using a finite-conjugate objective lens, I have read that you typically you want a 160mm tube-length.  Does the 50mm tube lens redirect the focus to make the finite-conjugate lens compatible with a 50mm tube?

Thanks again!

rwb27, 2018-02-15:
Thanks.  You might be confused, but you are also 100% correct!  The optics module you mention is correct for an RMS objective, the ThorLabs tube lens, and a Raspberry Pi camera module v2.  Looking at the Amazon link you included, I think you have 35mm parfocal distance lenses, so you'll need the LS65 (or LS65-M) body, and you won't need a riser on top of that, so you should also use the LS65 illumination module.

Any suggestions on how I could have explained that more clearly are appreciated - I think GitHub might even let you send me a pull request to update the readme file through the web interface.

Have fun, and do let me know how you get on - it is always lovely to hear people's build reports and what they use the microscope for!

rwb27, 2018-03-19:
Closing this because I think I answered the question - and you did manage to figure it out correctly from the instructions in the first place :)

closed 2018-03-19 by rwb27