rwb27/openflexure_microscope #70: main_body_LS65-M.stl

sh969, 2018-03-07:
New optics module mount:
It is a very complicated procedure to losen the screw each time I need to do modifications to the optics module (e.g. changing objective) since the centre gear (+motor) has to be unmounted as well. I much prefereed the old the clip-on solution.
The bended design of the centre actuator makes it very hard to attach its rubber band in the first place (even with new, longer tool).

Endstop screwholes:
The optical endstops are colliding with the z-actuator. It would help to move the screws a milimeter towards the outside (away from the centre).

rwb27, 2018-03-12:
Hello, I agree the new screw position isn't great.  There might be scope for a more complicated mechanism that puts the screw in a more accessible location, but I was waiting to check that the new z axis worked first.  The old clip-on solution works fine for a month or two, then the plastic loses its elasticity and the objective falls out the bottom of the microscope!  That's a bit of a problem, hence the change.  Using the new arrangement also results in a smaller, stiffer assembly.  I just need to fix the screw position and it will be lovely!

I should make a revised foot to fix the issue with band insertion, that oughtn't be too hard.

I can also shift the screw holes

sh969, 2018-04-06:
Successfully assembled the new version and I am quite happy with the stability of the optics module.

Two small issues:
1) The screwholes for the end-stops are still not quite fitting. I'ld suggest we move them ~4mm away from the optics module (towards the outside) and another ~3mm overhanging (there are pins under the endstop to connect them to the Arduino, so the sensor can only be mounted tilted as of now, see image 1). 
2) For some reason our 3d printers here in Maxwell (Ultimaker 2) are printing the optics module with a opening too small so the Raspberry Pi camera does not fit (even though the stl file looks fine). See red and yellow objective mount on image 2. The black one was printed somewhere else and looks how it is supposed to. Might be due to the fact that the cutout is on the bottom. I had to carve it out to make it fit. It was not easy at all to get rid of the extra material. Maybe just extend the opening depth in the design file.

[image]
[image]

rwb27, 2018-04-26:
We discussed the wierd opening and didn't get a solution - it seems quite hard to reproduce!

The issue with the difficult screw is (more or less) fixed in abb831b, which moves the access hole to the side, rather than above, the Z adjuster gear.

herbzhao, 2018-07-02:
The screw for the objective module takes some practice to install, but it works fine. However, is it possible to keep both push fit and screw? 

Holding the screw and optics module in place at the same time is difficult. Also, when loosening the screw to move the objective module, it may just slip down immediately.

rwb27, 2018-07-03:
It's hard to keep the push fit and the screw, in part because I am keen to make sure it self-centres, which I think would be difficult if I combined the dovetail and the screw.  If you can think of a way to make the old dovetail clip work with a screw I'd be interested - but I think it might be easier to change the new version slightly rather than reverting to the old one.

Currently, you can put the screw in, then twist the nut onto the end of the screw (usually just one turn, so there's plenty slack), then slide the optics module up onto the nut.  I think it might be easier if the nut is permanently mounted into the objective, and the screw can then be put on the objective, and inserted into the microscope through a slot.  I might have a go at doing that when I get a moment (!)

rwb27, 2018-07-25:
I think this is covered by other issues now - I'll close this one.

closed 2018-07-25 by rwb27