rwb27/openflexure_microscope #71: hybrid lens adapter for v1 AND v2 cams?

jywarren, 2018-03-09:
Hi! At Public Lab we're using the lens adapter part as part of a simpler "basic" microscope kit (a stepping stone to this project to help onboard newcomers) -- https://publiclab.org/notes/partsandcrafts/02-15-2018/2-attaching-your-raspberry-pi-camera-to-a-microscope-objective-lens

We notice that first-timers often mix up v1 and v2 cameras, and were wondering if anyone could post an stl for a lens adapter that would just fit either one -- that way we can skip a step and just smooth out the build process. Any help appreciated!

rwb27, 2018-03-10:
Hi - that's cool, I didn't realise you were using it for that.  Have you tried the latest version of the v2 optics (from the builds folder, [optics_picamera_2_pilens_SS40.stl](https://github.com/rwb27/openflexure_microscope/blob/master/builds/optics_picamera_2_pilens_SS40.stl))?  Because I've changed from push-fit to screw in (to help stop people destroying the flex cable) I *think* that should fit either, but it may be that the cut-out on the camera module is not quite right.  I don't have many v1 modules any more to test with, so if you could check that would help.  If it doesn't fit, it should be possible to tweak it fairly easily, so let me know and I'll see if I can come up with a universal version.  I think the lenses are similar enough in size that they should both fit the new mount, which is slightly flexible at the top.

jywarren, 2018-03-12:
Oh wow, thanks!!! 

Yes, i think there are some clearance issues, specifically in the two boxes I've drawn here:

[image]

The onboard camera ribbon attachment goes much closer to the edge of the adapter, so it needs half or more the width along that edge (at the top of this image).

There's also some tiny components (in the left-hand box region) that don't clear and I had to hollow out just a tiny bit of space for them. I'll post a picture of the board itself too.

jywarren, 2018-03-12:
Ah, here's a really good one, actually: 

[image]

steviepubliclab, 2018-04-12:
Hi! Also looking for modifications to the bottom piece to fit the Rev 1.3. This would be cutting out a window on the side opposite the screw holes. If you're holding it with the open side facing up, the widow needs to be towards the left hand side. 

See below:
[image]

[image]

[image]

rwb27, 2018-04-17:
Hi, I've taken a stab at this in the [unified picamera mount branch](https://github.com/rwb27/openflexure_microscope/tree/unified_picamera_mount/), specifically I've put up STLs in the builds folder for the three versions of the picamera optics module - e.g. the [SS40 version](https://github.com/rwb27/openflexure_microscope/blob/unified_picamera_mount/builds/optics_picamera_2_pilens_SS40.stl) that uses the pi lens.

[image]

I've yet to test this out - if anyone interested fancies printing one and checking, that would be super.

rwb27, 2018-04-17:
I'm going to leave it called "picamera_2" for now, because the lens removal tools are still specific to that version.  I can probably find an STL for the board gripper that works with v1, but I never did manage to make my lens removal tool work with v1, it's v2 only.  I'm also surprised there are still people building with v1 modules - I thought they were discontinued?

jywarren, 2018-04-18:
We're ordering lots of v1.3 modules via Alibaba and Aliexpress -- pretty good, and much cheaper! 

Thanks so much for this adaptation, @rwb27 -- we're definitely going to test it out and will get back to you. 

Much appreciated! 👍 🙌 🎉

rwb27, 2018-04-19:
no problem.  If you're able to measure where the window needs to be in the back part, it should be easy enough to add a modification in - even better, you might be able to do it directly in OpenSCAD and send me a pull request - https://github.com/rwb27/openflexure_microscope/blob/master/openscad/cameras/picamera_2_cover.scad will generate the cover as you have it now, so you'd just need to take the difference of the current cover and a rectangle that chops out the relevant bit, and call it picamera_1_cover.scad.

rwb27, 2018-07-05:
@jywarren did this work for you in the end?

rwb27, 2018-08-09:
I've pulled this into both the master branch and my current development branch - I will close the issue for now but please do re-open it if you figure out how to modify the cover, and do post back to say if the hybrid part works for you :)

jywarren, 2018-08-09:
Hi, @rwb27 -- sorry, ok, we'll test this one -- https://github.com/rwb27/openflexure_microscope/blob/master/builds/optics_picamera_2_pilens_SS40.stl

If that works, have those geometries been adapted into the RMS designs as well? Thanks and sorry it took us a while to test this out!

jywarren, 2018-08-09:
This commit seems to indicate so! https://github.com/rwb27/openflexure_microscope/commit/7eefc52800669a2fc5e880acf5dff288f766a058#diff-d862a606811deb350fed830c41b0f0f2

jywarren, 2018-08-09:
Looking at https://github.com/rwb27/openflexure_microscope/blob/4eecaadd426169f551a0be46a430c6b15da4e1c7/builds/optics_picamera_2_rms_f50d13_LS65.stl and https://github.com/rwb27/openflexure_microscope/blob/4eecaadd426169f551a0be46a430c6b15da4e1c7/builds/optics_picamera_2_rms_f40d16_LS65.stl

rwb27, 2018-08-10:
hmm, looking at the STLs I think that might not be the case - only the SS40 one looks right.  If you could check that one actually fits, it oughtn't be a big deal to recompile the others.

jywarren, 2018-08-30:
Hmm, actually all three fit! 

https://github.com/rwb27/openflexure_microscope/blob/master/builds/optics_picamera_2_pilens_SS40.stl

and:

https://github.com/rwb27/openflexure_microscope/blob/4eecaadd426169f551a0be46a430c6b15da4e1c7/builds/optics_picamera_2_rms_f50d13_LS65.stl

https://github.com/rwb27/openflexure_microscope/blob/4eecaadd426169f551a0be46a430c6b15da4e1c7/builds/optics_picamera_2_rms_f40d16_LS65.stl

They all three seem to have the same cutouts in the base:

[image]
[image]

rwb27, 2018-08-30:
Yes, I think I recompiled them all after I realised that. They are definitely all the new universal cutout :)

jywarren, 2018-08-30:
fantastic, and thanks so much!

rwb27, 2019-01-14:
By the way, if you are not using the dovetail on the side to mount them, you might want to check out the new "lens spacer tube" that I'm using in the upcoming release - this is the part you see above but without the mounting.  It's designed to screw on top of the camera, mounting to four M2 holes in a baseplate.  I'll close this issue for now, but do shout if there's anything further I can do to be helpful :)

closed 2019-01-14 by rwb27