rwb27/openflexure_microscope #83: Importance of the elastic bands for the motorized setup

fanneaup, 2018-07-24:
I reproduced the openflexure (SS40 with motors) microscope and adapted it to work in reflection.

However, I am having trouble getting smooth, repeatable motion from the motors. I was wondering how important the type and size of elastic bands was or if anyone had come across the same issue.

rwb27, 2018-07-24:
The elastic bands are generally pretty important - assuming you're using the latest version, you should be using the 30mm inner diameter, 2mm thickness Viton O rings.  These can be ordered from e.g. simplybearings.co.uk.  Using a band that's too strong will cause the stage to stick while it moves, while a weak one will mean that for a large part of its travel, the microscope flops around erratically.

Your choice of screw/nut can also have an effect; brass nuts on stainless steel screws with a little grease is my preferred option.  What are you using for the motors?

If you upload a picture or two here (you should be able to drag and drop them) I'll have a look - sometimes there's something obvious in the picture.

fanneaup, 2018-07-25:
[image]
[image]

Thank you for your answer.

Pictures of my setup are attached. The motors are 28BYJ-48 with their ULM 2003 driver boards. they are controlled by the GPIO pins of the Raspberry Pi 3 B that I am also using for the microscope but they are alimented by an external power supply.

When it comes to the screws and nuts, both are made of steel and I have put some WD40 as lubricant.

Eventually, the elastic bands seem to be 2x1mm width and 50mm of diameter but I have cut them to make them thiner and tighter.

rwb27, 2018-07-25:
ok, looks like you've assembled it nicely, and you have the current master branch version - excellent.  The rubber bands will most probably be less strong than the Viton O rings, given that you're only in Bristol I can probably get some O rings to you relatively easily.  The best way to test if the bands are doing their job is to unscrew the large gear until there is a visible space between the gear and the column it sits on.  You should then look at the bottom of the microscope, and check that the actuator has been pulled down - I think the tip of the actuator should move about 4mm down, so the edge of the wider part moves by something like 2-3mm.  If it's not protruding down as much as that, the band is too loose/not strong enough.

I have never used the GPIO pins on the Raspberry Pi to control the motors, so I don't know how good the timing is; are you using half-stepping, and how fast are you stepping the motors?  I would usually trust an Arduino more than the Pi GPIO pins, because its timing is better.  If you can get hold of a Nano (or equivalently an Uno) you could use the [[rwb27/openflexure_nano_motor_controller]] firmware (and Python module).

fanneaup, 2018-07-25:
Ok, I just did what you said ("unscrew the large gear until there is a visible space between the gear and the column it sits on") and it looks like this: 

[image]

So the edge of the wider part moved down closer of 1mm than 3mm. I can make it move down to 3mm by tighening the elastic bands again but then the stepper motors seem to be blocked by the elastic and are not able to move.

In order to rotate the stepper motors, I am currently using half-stepping with a delay of 0.003s (value chosed according to the datasheet). But, at the beginning, I had tighter elastic bands that did a great job on the actuators but I thought they were to strong for the motors (as they were stuck and enable to rotate) so I tried normal stepping with the same delay as I had read it would give them more torque but it was unsuccessful. To fix that, I have loosened the elastic bands to allow my motors to rotate but maybe it came from the timing of the Pi as you said because it does not give me a smooth motion.

rwb27, 2018-07-25:
hmm, ok.  The picture you've pasted in up there looks about right - are you sure that's only 1mm?  I'd say the elastic bands are probably fine as you have them.

It may well be worth trying an arduino, if the Pi is skipping steps that would explain it.  I have also had issues with one (or more) of the pins driving the motor being disconnected - this often shows up as a drop in torque.

Lastly, I tend to go for grease rather than something like WD40; it lasts longer and results in smoother motion.  Brass nuts are also worth getting hold of - it feels noticeably smoother.

voxelman, 2018-11-09:
WD40 is not a particularly good lubricant for threads because it evaporates. It is actually a water displacement solution. It would be better to use a thread lubricant grease. For that I would say that Google and Amazon are your friend.

rwb27, 2018-11-12:
thanks - my current favourite is either sewing machine oil (very light, wicks well into the M3 threads) or 3-in-1 (also fairly light, and easier to find).  That said, I think what the grease was mostly doing is lubricating the washer on top (between the plastic gear and the top of the microscope.  I find that two washers + grease makes the biggest difference, and the next biggest improvement comes from using decent steel screws (I have some cheap screws that strip after a couple of days of heavy use).  Once you've got two washers and a decent screw, an appropriate grease/oil is definitely important, and I should add that to the instructions.

rwb27, 2018-11-12:
By the way @fanneaup could you post some details of how you're using it in reflection?  It's something a number of people are interested in, and I'd love to see how you've done it (and how well it worked).

rwb27, 2019-01-21:
This thread has gone quiet for a while - now that I have the new instructions out (which are clearer about the Viton O rings being the preferred option) I think it can probably be closed - but please do re-open it if there's more to add, or if you think the instructions could be improved!

closed 2019-01-21 by rwb27