rwb27/openflexure_microscope #88: Mesh repair needed on main body

rwb27, 2018-10-03:
I've just started using Slic3r again, and noticed that there are some bad edges/faces in the STL of the main body in the cantilever branches.  I should track this down and fix it, as I think it causes printing issues on my Prusa, which is a shame!  It's probably a matter of looking back through old commits and finding the last STL that doesn't report any mesh errors, then looking at the SCAD files to figure out what changed.  Figuring out which STL is the last valid one shouldn't be too tricky, so would be a great way for someone to get involved without needing to learn OpenSCAD, and would save us a load of time...

voxelman, 2018-10-06:
I’ve printed the main body stepper motor version from the master in the past two weeks on my Original Prusa i3 MK2.5 without any complaints from the PE version of Slic3r. If the Slic3r fixed problems, I didn’t notice. I will check it tomorrow and let you know if I notice any irregularities. Thank you for the great work.

rwb27, 2018-12-17:
Hi @voxelman thanks for posting that - I think the version on master is OK, it was the version in the cantilevered_objective_mount branch that caused problems.  I'm now working on "reversed_cantilevered_objective_mount", and that seems to be OK again in Slic3r, though I am not sure what fixed it!  I am working on tidying up the branches so the development branch is more obvious...

rwb27, 2018-12-17:
as of f70bd71075a49a5887644b359ecbd21003f4c3e8 it seems to be valid, though reports as two shells rather than one.

rwb27, 2019-01-16:
I thought I'd tracked this down in OpenSCAD, because it was reporting that there were 2 volumes - but it seems that OpenSCAD always reports 2 volumes.  I came across a mailing list thread suggesting that CGAL consders the shape to be one volume, and the infinite space outside of the shape to be the second volume - so there are always two.

rwb27, 2019-01-16:
I've also had a closer look at the output from Slic3r; it looks fine in a G code viewer, but what my Mk3 is doing doesn't correspond to the contents of the G code file, I think maybe it has skipped some lines.  This is wierd, but I am starting to think it might be our printer and not the model.

rwb27, 2019-01-16:
Oh, and according to MakePrintable.com the LS65-M main body has "no issues" (although it does mention 29 non-manifold edges).

[image]

rwb27, 2019-01-18:
I have found the (current) problem - it was my printer!  The G code and STL were fine, I power cycled the Prusa and the body is printing fine now.  It's wierd that the errors were so consistent, maybe there's a particular combination of moves that causes it to skip stuff (like a curve with many short segments followed by a long line)?  Anyway, I think the STL is valid so I'm closing the issue :)

closed 2019-01-18 by rwb27