rwb27/openflexure_microscope #97: base plate

paulnyakyi, 2019-01-24:
could we have a short baseplate that doesn`t accommodate raspberry pi?for  enclosure setup that we can build in future

rwb27, 2019-01-24:
I think you actually have this already, in the low cost microscope design - I must have made it and then forgotten!  I have added it back in, as microscope_stand_no_pi.stl (in the builds folder) and the OpenSCAD file with the same name, in 05587bc892eda9099caf378937ad5381ca8d99ef.  You will probably need to add cut-outs to it for the cables, perhaps Grace can do this in OpenScad?

codeTom, 2019-02-13:
This is the base we used for the low cost version last summer: 
https://github.com/codeTom/openflexure_microscope/blob/662a3e987d1c8331aa8ca6efb110befea33f4a9c/openscad/microscope_stand.scad

rwb27, 2019-04-30:
The repository is [moving to GitLab](https://gitlab.com/openflexure/openflexure-microscope/), and this issue has been migrated.  I'll close all the issues here in due course, but am leaving notices on all the currently-open ones.  If you head over to the other repository, this issue will be updated there.