rwb27/openflexure_microscope #101: http://rwb27.github.io/openflexure_microscope/docs/parts/hires_optics_casing gives 404 error

ctzurcanu, 2019-01-29:
when clicked from page:
http://rwb27.github.io/openflexure_microscope/docs/2b_high_resolution_optics_module.html

rwb27, 2019-01-29:
Thanks - most of the parts links currently don't work - if you check section 0 - printing, that's probably the most reliable place to work stuff out from.  Currently, the link above is a single printed part, the one that starts "optics_...stl" (see the printing section to figure out which version you need, I suspect it's probably optics_picamera_2_rmsf50d13_LS65.stl

rwb27, 2019-01-29:
I'll leave this open, to remind me to update the parts links - my only reluctance to sink a lot of effort into that is that I'm really hoping we'll automate that part of the documentation soon...  Please do add broken links here, though anything that starts docs/parts is likely to be missing.

rwb27, 2019-04-24:
I'm slowly populating the parts folder - as of 8562c72be3edd436e9ea222f651bf2d1bf97926b I've got many of the non-printed parts linking through from the bill of materials page - need to propagate those to the instructions, though.

rwb27, 2019-04-30:
The repository is [moving to GitLab](https://gitlab.com/openflexure/openflexure-microscope/), and this issue has been migrated.  I'll close all the issues here in due course, but am leaving notices on all the currently-open ones.  If you head over to the other repository, this issue will be updated there.