rwb27/openflexure_microscope #102: Build report

ctzurcanu, 2019-01-29:
First: I have discovered this project quite late for my purposes. I realized that I may need quite a long time to gather all the required pieces. (I have only one month to complete this and some parts require 2-3 months to arrive from China)

One example: 
https://www.aliexpress.com/item/High-power-LED-lens-diameter-13mm-Plano-Convex-lens-led-optical-lens-1W-3W-Reflector-Lens/1262190515.html?spm=2114.search0104.3.2.27c1304cILXAPH&ws_ab_test=searchweb0_0,searchweb201602_1_10320_10065_10068_10547_319_317_10696_453_10084_454_10083_433_10618_431_10304_10307_10820_10301_10821_537_536_10902_10843_10059_10884_10887_100031_321_322_10103,searchweb201603_59,ppcSwitch_0&algo_expid=dcb64a9e-830f-4a04-93d6-3106ed6a4c3a-0&algo_pvid=dcb64a9e-830f-4a04-93d6-3106ed6a4c3a

or:
https://www.aliexpress.com/item/4X-10X-40X-100X-Infinity-Plan-Achromatic-Objective-Lens-with-RMS-Mounting-Thread-DIN-45mm-f/32912050210.html?spm=2114.search0104.3.46.5df06a99vZWRd1&ws_ab_test=searchweb0_0,searchweb201602_1_10320_10065_10068_10547_319_317_10696_453_10084_454_10083_433_10618_431_10304_10307_10820_10301_10821_537_536_10902_10843_10059_10884_10887_100031_321_322_10103,searchweb201603_59,ppcSwitch_0&algo_expid=031a16ae-a830-45fa-8a77-18dad23fe003-7&algo_pvid=031a16ae-a830-45fa-8a77-18dad23fe003

But I will try to press ahead with the build.

I have ordered:
https://kitspace.org/boards/github.com/rwb27/openflexure_nano_motor_controller/
3x PCB
3x the rest of electronics minus the ICs, because the ICs will come with the motors:

https://www.amazon.de/gp/product/B0711TWN12/ref=ppx_yo_dt_b_asin_title_o00__o00_s00?ie=UTF8&psc=1

rwb27, 2019-01-29:
Thanks for starting a build report.

Regarding the optical bits, it's worth pointing out that the basic version works fine without the plastic lens (you can experiment with a piece of white tape or thin paper where the lens should go, sometimes that acts nicely as a diffuser).  That is one reason I tend to encourage people to build the basic version first.

Sourcing objectives and lenses from China is generally a bit slow, but suppliers in the UK (and I suspect many other places) are several times more expensive.  Objective lenses can be obtained from e.g. ThorLabs, Edmund Optics, Comar Optics, etc. but usually at a significantly higher price.  They tend to arrive next-day, though!

I've not found a good UK supplier of the condenser lens, more expensive glass ones are available from the companies above, but this is way more than we need for the OpenFlexure Microscope.

ctzurcanu, 2019-02-01:
Waiting for Godot...

I have most of what I need for 3 microscopes (all motorized)

[image]

Waiting for the 3 PCBs and the 3D printed parts.

ctzurcanu, 2019-02-06:
Half of Godot is here!
[image]

This is only a part of the printed parts :)
Used https://www.3dhubs.com/ for printing. The quality seems good.

ctzurcanu, 2019-02-06:
other small parts:
[image]

Now we can begin the assembly!

ctzurcanu, 2019-02-16:
Godot arrived! (the PCBs arrived, we soldered them)
Now having a setup problem: one time the install script worked well - now it does not more at #106 

[image]

ctzurcanu, 2019-02-16:
This software problem is the only one obstacle remaining. Now we are at home and able to find the last screws needed. 
The saddening part is that we made it to work couple of days ago. We played with the motors remotely. Now it seems to fail at the exact (past) success formulae.

ctzurcanu, 2019-02-16:
The microscope objective that I have here at home seems to fit the bill. 100x on oil. Looks like this:
[image]

ctzurcanu, 2019-02-19:
I have gathered all the needed bits. Here is one completed hi-res micro:

[image]

rwb27, 2019-02-20:
Well done, that looks like it's really nicely assembled!  The two base parts look identical though, normally the top one would be a different shape, to allow a bit more clearance for the optics module, and so that there are screw holes that match with the lugs on the micorscope.  Did you print the microscope_stand part, or two copies of the motor driver casing?

rwb27, 2019-02-20:
(it's worth pointing out that it should work either way, it's just that the version that screws together is a bit more robust).

ctzurcanu, 2019-02-20:
Ha ha ha! You caught me: I had to change and put the rasp-specific component first because this is the hi-res montage. Yes: optics needed more room to go down.

The parts need a bit of modification and would be truly interchangeable. with longer screws, one could screw directly into 2 of the feet.

ctzurcanu, 2019-02-20:
[image]
One first slide. this is lo-res

rwb27, 2019-02-20:
very nice :)

rwb27, 2019-04-30:
The repository is [moving to GitLab](https://gitlab.com/openflexure/openflexure-microscope/), and this issue has been migrated.  I'll close all the issues here in due course, but am leaving notices on all the currently-open ones.  If you head over to the other repository, this issue will be updated there.

closed 2019-04-30 by rwb27