rwb27/openflexure_microscope #106: Using the install script stalls flask

ctzurcanu, 2019-02-15:
pi@raspberrypi:~ $ systemctl status ofmserver

ofmserver.service - Gunicorn instance to serve the OpenFlexure Microscope API
   Loaded: loaded (/etc/systemd/system/ofmserver.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-02-15 23:41:39 EET; 3min 55s ago
 Main PID: 2265 (gunicorn)
   CGroup: /system.slice/ofmserver.service
           ├─2265 /home/pi/.openflexure/envmicroscope/bin/python3 /home/pi/.openflexure/envmicroscope/bin/gunicorn --threads 5 --workers 1 --graceful-timeout 3 --bind 0.0.0.0:5000 openflexure_microscope.api.app:app
           └─2270 /home/pi/.openflexure/envmicroscope/bin/python3 /home/pi/.openflexure/envmicroscope/bin/gunicorn --threads 5 --workers 1 --graceful-timeout 3 --bind 0.0.0.0:5000 openflexure_microscope.api.app:app

Feb 15 23:42:22 raspberrypi gunicorn[2265]: DEBUG:root:Setting parameter image_resolution: [2592, 1944]
Feb 15 23:42:22 raspberrypi gunicorn[2265]: DEBUG:root:Entering worker thread.
Feb 15 23:42:22 raspberrypi gunicorn[2265]: INFO:root:Waiting for frames
Feb 15 23:42:22 raspberrypi gunicorn[2265]: DEBUG:root:Started MJPEG stream at [832, 624] on port 1
Feb 15 23:42:22 raspberrypi gunicorn[2265]: DEBUG:root:STREAM ACTIVE
Feb 15 23:42:22 raspberrypi gunicorn[2265]: DEBUG:root:Creating stage object...
Feb 15 23:42:42 raspberrypi gunicorn[2265]: INFO:root:Stopped MJPEG stream on port 1. Switching to [2592, 1944].
Feb 15 23:42:42 raspberrypi gunicorn[2265]: DEBUG:root:FRAME ITERATOR END
Feb 15 23:42:42 raspberrypi gunicorn[2265]: DEBUG:root:BaseCamera worker thread exiting...
Feb 15 23:42:42 raspberrypi gunicorn[2265]: DEBUG:root:Stream thread is None

This was done on a fresh raspberry pi:

pi@raspberrypi:~ $ curl -sS https://gitlab.com/openflexure/openflexure-microscope-installer/raw/master/openflexure_microscope.sh |sudo bash

This script will install everything needed to use
OpenFlexure Microscope

Always be careful when running scripts and commands copied
from the internet. Ensure they are from a trusted source.

If you want to see what this script does before running it,
you should run: 'curl https://gitlab.com/openflexure/openflexure-microscope-installer/raw/master/openflexure_microscope.sh'

Running as root, please log in as a regular user with sudo rights!

Do you wish to continue? [Y/n] y
Working in /home/pi

Checking environment...

Installing core dependencies...
libatlas-base-dev is already installed
libjasper-dev is already installed
libjpeg-dev is already installed

Install into a virtual environment? [Y/n] y
Requirement already satisfied: virtualenv in /usr/local/lib/python3.5/dist-packages
Virtualenv directory already exists. Skipping to activation.
Activating virtualenv...
/home/pi/.openflexure/envmicroscope/bin/python3
/home/pi/.openflexure/envmicroscope/bin/pip3

Checking for packages required for GPIO control...
raspi-gpio is already installed
Installing Python RPi.GPIO
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: RPi.GPIO in ./.openflexure/envmicroscope/lib/python3.5/site-packages (0.6.5)
RPi.GPIO installed and up-to-date

Installing OpenFlexure Microscope
Install in developer mode? [y/N] y
Retrieving Poetry metadata

# Welcome to Poetry!

This will download and install the latest version of Poetry,
a dependency and package manager for Python.

It will add the `poetry` command to Poetry's bin directory, located at:

$HOME/.poetry/bin

This path will then be added to your `PATH` environment variable by
modifying the profile file located at:

$HOME/.profile

You can uninstall at any time with `poetry self:uninstall`,
or by executing this script with the --uninstall option,
and these changes will be reverted.

Installing version: 0.12.11
  - Downloading poetry-0.12.11-linux.tar.gz (8.13MB)

Poetry (0.12.11) is installed now. Great!

To get started you need Poetry's bin directory ($HOME/.poetry/bin) in your `PATH`
environment variable. Next time you log in this will be done
automatically.

To configure your current shell run `source $HOME/.poetry/env`

Creating /home/pi/openflexure-microscope-server
Cloning into '/home/pi/openflexure-microscope-server'...
remote: Enumerating objects: 283, done.
remote: Counting objects: 100% (283/283), done.
remote: Compressing objects: 100% (166/166), done.
remote: Total 2052 (delta 159), reused 211 (delta 116)
Receiving objects: 100% (2052/2052), 357.72 KiB | 628.00 KiB/s, done.
Resolving deltas: 100% (1265/1265), done.

Poetry-installing openflexure-microscope. This may take a while!
Installing dependencies from lock file

Package operations: 33 installs, 0 updates, 0 removals

  - Installing certifi (2018.11.29)
  - Installing chardet (3.0.4)
  - Installing idna (2.8)
  - Installing markupsafe (1.1.0)
  - Installing pyparsing (2.3.0)
  - Installing pytz (2018.9)
  - Installing six (1.12.0)
  - Installing urllib3 (1.24.1)
  - Installing alabaster (0.7.12)
  - Installing babel (2.6.0)
  - Installing click (7.0)
  - Installing docutils (0.14)
  - Installing imagesize (1.1.0)
  - Installing itsdangerous (1.1.0)
  - Installing jinja2 (2.10)
  - Installing packaging (18.0)
  - Installing pygments (2.3.1)
  - Installing requests (2.21.0)
  - Installing snowballstemmer (1.2.1)
  - Installing sphinxcontrib-websupport (1.1.0)
  - Installing werkzeug (0.14.1)
  - Installing flask (1.0.2)
  - Installing future (0.17.1)
  - Installing numpy (1.15.4)
  - Installing pyserial (3.4)
  - Installing sphinx (1.8.3)
  - Installing flask-cors (3.0.7)
  - Installing openflexure-stage (0.2.1)
  - Installing picamera (1.13)
  - Installing pillow (5.4.1)
  - Installing pyyaml (3.13)
  - Installing scipy (1.2.0)
  - Installing sphinxcontrib-httpdomain (1.7.0)
  - Installing openflexure_microscope (1.0.0-beta.7)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting gunicorn
  Downloading https://files.pythonhosted.org/packages/8c/da/b8dd8deb741bff556db53902d4706774c8e1e67265f69528c14c003644e6/gunicorn-19.9.0-py2.py3-none-any.whl (112kB)
    100% |████████████████████████████████| 122kB 1.2MB/s 
Installing collected packages: gunicorn
Successfully installed gunicorn-19.9.0
Install unofficial picamera supporting lens-shading? [Y/n] y
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting picamera==1.13.1b0 from https://github.com/rwb27/picamera/releases/download/v1.13.1b0/picamera-1.13.1b0-py3-none-any.whl
  Downloading https://github.com/rwb27/picamera/releases/download/v1.13.1b0/picamera-1.13.1b0-py3-none-any.whl (161kB)
    100% |████████████████████████████████| 163kB 668kB/s 
Installing collected packages: picamera
  Found existing installation: picamera 1.13
    Uninstalling picamera-1.13:
      Successfully uninstalled picamera-1.13
Successfully installed picamera-1.13.1b0

Installing microscope service
Start the server on boot? [Y/n] y
Created symlink /etc/systemd/system/multi-user.target.wants/ofmserver.service → /etc/systemd/system/ofmserver.service.
Start the server now? [Y/n] y

Install the local keyboard control client? [Y/n] y
Creating /home/pi/openflexure-microscope-keyclient
Cloning into '/home/pi/openflexure-microscope-keyclient'...
remote: Enumerating objects: 44, done.
remote: Counting objects: 100% (44/44), done.
remote: Compressing objects: 100% (37/37), done.
remote: Total 44 (delta 19), reused 0 (delta 0)
Unpacking objects: 100% (44/44), done.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting readchar (from -r /home/pi/openflexure-microscope-keyclient/requirements.txt (line 1))
  Downloading https://files.pythonhosted.org/packages/ef/f3/50cc9199733bf91f3c140150752a6430d8b785f9d22a31dcae721edb0b9a/readchar-2.0.1-py3-none-any.whl
Requirement already satisfied: requests in ./.openflexure/envmicroscope/lib/python3.5/site-packages (from -r /home/pi/openflexure-microscope-keyclient/requirements.txt (line 2)) (2.21.0)
Requirement already satisfied: urllib3<1.25,>=1.21.1 in ./.openflexure/envmicroscope/lib/python3.5/site-packages (from requests->-r /home/pi/openflexure-microscope-keyclient/requirements.txt (line 2)) (1.24.1)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in ./.openflexure/envmicroscope/lib/python3.5/site-packages (from requests->-r /home/pi/openflexure-microscope-keyclient/requirements.txt (line 2)) (3.0.4)
Requirement already satisfied: idna<2.9,>=2.5 in ./.openflexure/envmicroscope/lib/python3.5/site-packages (from requests->-r /home/pi/openflexure-microscope-keyclient/requirements.txt (line 2)) (2.8)
Requirement already satisfied: certifi>=2017.4.17 in ./.openflexure/envmicroscope/lib/python3.5/site-packages (from requests->-r /home/pi/openflexure-microscope-keyclient/requirements.txt (line 2)) (2018.11.29)
Installing collected packages: readchar
Successfully installed readchar-2.0.1

Finished!

ctzurcanu, 2019-02-16:
The manual install fails in the same way: flask would not return any request.

The log follows:
```
(env) pi@raspberrypi:~/openflexure-microscope-server $ gunicorn --threads 5 --workers 1 --graceful-timeout 3 --bind 0.0.0.0:5000 openflexure_microscope.api.app:app
[2019-02-16 21:41:05 +0200] [2440] [INFO] Starting gunicorn 19.9.0
[2019-02-16 21:41:05 +0200] [2440] [INFO] Listening at: http://0.0.0.0:5000 (2440)
[2019-02-16 21:41:05 +0200] [2440] [INFO] Using worker: threads
[2019-02-16 21:41:05 +0200] [2443] [INFO] Booting worker with pid: 2443
DEBUG:root:Initialising /home/pi/.openflexure/microscoperc.yaml
DEBUG:root:Exists: True
INFO:root:Loading /home/pi/.openflexure/microscoperc.yaml...
DEBUG:root:Expanding ~/.openflexure/picamera_settings.yaml
DEBUG:root:Initialising /home/pi/.openflexure/picamera_settings.yaml
DEBUG:root:Exists: True
INFO:root:Loading /home/pi/.openflexure/picamera_settings.yaml...
DEBUG:root:Attaching camera...
INFO:root:Attached dummy camera.
DEBUG:root:Attaching stage...
INFO:root:Attached dummy stage.
Creating plugin mount
INFO:root:Reading config directly
INFO:root:Reading config directly
DEBUG:root:Attaching plugins...
DEBUG:root:Checking openflexure_microscope
DEBUG:root:Checking openflexure_microscope.plugins
DEBUG:root:Checking openflexure_microscope.plugins.default
DEBUG:root:Checking openflexure_microscope.plugins.default.autofocus
['openflexure_microscope', 'plugins', 'default', 'autofocus']
INFO:root:Plugin openflexure_microscope.plugins.default.autofocus:AutofocusPlugin loaded as default/autofocus.
DEBUG:root:Checking openflexure_microscope
DEBUG:root:Checking openflexure_microscope.plugins
DEBUG:root:Checking openflexure_microscope.plugins.default
DEBUG:root:Checking openflexure_microscope.plugins.default.camera_calibration
['openflexure_microscope', 'plugins', 'default', 'camera_calibration']
INFO:root:Plugin openflexure_microscope.plugins.default.camera_calibration:Plugin loaded as default/camera_calibration.
DEBUG:root:Plugins attached.
INFO:root:Reading config directly
INFO:root:Reading config directly
DEBUG:root:Registering blueprints...
DEBUG:root:Created app route: /api/v1
DEBUG:root:Created app route: /api/v1/stage
DEBUG:root:Created app route: /api/v1/camera
DEBUG:root:/default/autofocus/autofocus
DEBUG:root:/default/autofocus/measure_sharpness
DEBUG:root:/default/camera_calibration/recalibrate
DEBUG:root:Created app route: /api/v1/plugin
DEBUG:root:Created app route: /api/v1/task
DEBUG:root:First request made. Populating microscope with hardware...
DEBUG:root:Creating camera object...
INFO:root:Reading config directly
INFO:root:setting zoom, centre [0.5 0.5], size 1.0
DEBUG:root:Applying config:
DEBUG:root:{'plugins': ['openflexure_microscope.plugins.default.autofocus:AutofocusPlugin', 'openflexure_microscope.plugins.default.camera_calibration:Plugin'], 'picamera_settings': {}, 'jpeg_quality': 75, 'numpy_resolution': [1312, 976], 'name': 'cdbc6ad019874766ab1cb0744059a971', 'image_resolution': [2592, 1944], 'fov': [4100, 3146], 'video_resolution': [832, 624]}
DEBUG:root:Setting parameter plugins: ['openflexure_microscope.plugins.default.autofocus:AutofocusPlugin', 'openflexure_microscope.plugins.default.camera_calibration:Plugin']
DEBUG:root:Setting parameter jpeg_quality: 75
DEBUG:root:Setting parameter numpy_resolution: [1312, 976]
DEBUG:root:Setting parameter name: cdbc6ad019874766ab1cb0744059a971
DEBUG:root:Setting parameter image_resolution: [2592, 1944]
DEBUG:root:Setting parameter fov: [4100, 3146]
DEBUG:root:Setting parameter video_resolution: [832, 624]
DEBUG:root:Entering worker thread.
INFO:root:Waiting for frames
DEBUG:root:Started MJPEG stream at [832, 624] on port 1
DEBUG:root:STREAM ACTIVE
DEBUG:root:Creating stage object...
INFO:root:Stopped MJPEG stream on port 1. Switching to [2592, 1944].
DEBUG:root:FRAME ITERATOR END
DEBUG:root:BaseCamera worker thread exiting...
DEBUG:root:Stream thread is None
```

ctzurcanu, 2019-02-16:
with the mention that it stayed at least 30 secs at
`DEBUG:root:Creating stage object...`

ctzurcanu, 2019-02-16:
I updated the firmware and it works now. Maybe the usb was not plugged well...
The `BasicSerialInstrument.query` was blocking the thread.

closed 2019-02-16 by ctzurcanu

rwb27, 2019-02-18:
Ah, sorry - I thought I'd fixed this, but maybe there is an old copy of the firmware kicking around; the software can hang if the motor board is not plugged in, or running out of date/wrong firmware.  I should probably add a timeout and a helpful error message, that would save some head-scratching.

By the way, if you have software-related issues, it's probably best to raise an issue on GitLab.com/openflexure/openflexure-microscope-server (or the installer repo if that's the issue) as that will reach Joel more directly.  There is a software repository on Github, but we are gradually migrating to the (rewritten and much better) version on GitLab.

ctzurcanu, 2019-02-18:
I have to tell you the story then:
I had to re-solder a capacitor and probably the board lost the firmware. But I think the software should catch that. I lost about a week scratching my stupid head...