rwb27/openflexure_microscope #107: not enough power to motors? maybe I need to power the driver PCB separately?

ctzurcanu, 2019-02-16:
When I try different values for x,y,z I noticed that for 500 the motors rotate. But for -500 they do not. When returning to 500, they work again.
Power is from the Arduino nano power that is into Rasp PI board.

rwb27, 2019-02-18:
Ah, yes - I always use separate motor power.  I think the v2 Sangaboard does have an unfortunate connection between the 5v pin on the Arduino and the 5v motor power - we have had issues with back-powering the Raspberry Pi through the USB line, you may want to think about cutting the trace that connects to the Arduino's 5v pin...

My usual method is to put screw terminals on the Sangaboard, and then wire a 5v Raspberry Pi into those, and then wire from the screw terminals into the Pi's micro USB power port (using the cut-off end of the cable).  If I'm being slightly nicer, I use a male-female micro USB lead, and cut it in the middle to expose the 5v and 0v lines, then do the same thing.  It's a bit of a bodge; the v3 Sangaboard has both micro USB power input and USB type A power output for the Pi, but it's currently not in the repo (I do periodically remind @sangavalerian to upload it).  It's surface-mount only though, and requires a 2 sided board with vias.  We're getting the first batch made by Seeed now.

rwb27, 2019-02-20:
I think you commented in the build report #102 that you used the power directly from the Pi (I guess the 0v and 5v GPIO pins?).  I've not usually done this as I was worried about blowing the polyfuse on the Pi, so it's interesting if that works for you...

ctzurcanu, 2019-02-20:
I figured that if Pi remains in server mode, it consumes only about 5% CPU. It should not draw a lot of juice. but I have not measured.
Yes. It is risky and I may take the power from the source, but I did not have sockets home now to use.
Maybe in 2-3 days, I will get to this too.

rwb27, 2019-04-30:
did you get this working in the end?  I think from your build report you might have done?  I should have added that my usual solution to having too few sockets is to make a custom power lead, with one micro-USB and another connector that I can plug directly in to the motor board.  One 10W Raspberry Pi PSU is usually sufficient, but this way avoids the polyfuse.

rwb27, 2019-04-30:
The repository is [moving to GitLab](https://gitlab.com/openflexure/openflexure-microscope/), and this issue has been migrated.  I'll close all the issues here in due course, but am leaving notices on all the currently-open ones.  If you head over to the other repository, this issue will be updated there.