rwb27/openflexure_microscope #109: Whaat way to insert the lens

ctzurcanu, 2019-02-18:
at: http://rwb27.github.io/openflexure_microscope/docs/2a_basic_optics_module.html

Step 7: It is not clear which side of the lens should face inside the dark chamber.

rwb27, 2019-02-18:
thanks for flagging that up, I should definitely update the instructions.  The side that normally faces the camera sensor should face the sample, so the side that normally faces out (with the smaller window, and the four flanges that you used to unscrew it) should face down.

rwb27, 2019-02-27:
I just updated the text in step 7 and I think it's fairly clear (ca77ff84245f3d23c2e92ca1d59a6f1fd2a01a81) - I guess a picture showing the right and wrong way to orient the lens would be helpful though.

rwb27, 2019-04-30:
The repository is [moving to GitLab](https://gitlab.com/openflexure/openflexure-microscope/), and this issue has been migrated.  I'll close all the issues here in due course, but am leaving notices on all the currently-open ones.  If you head over to the other repository, this issue will be updated there.