rwb27/openflexure_microscope #112: What lenses do I need to use a 10x NPL FLUOTAR Objective?

rwb27, 2019-03-13:
I want to acquire tiled fluorescence images of 'comet cell' assays, so don't need huge magnifications, and we use SYBR safe for green fluorescence.  I have an NPL FLUOTAR 10x objective available (0.3) from Leitz, and wonder for the optics parts, if that will be sufficient. However, is it the piece f40d16 or the f50d13 that would be the most appropriate to make?  It is clear that one is shorter than the other, of course...  (next step before printing - to see if our black filament is matt or shiny!) . Thanks!

_Originally posted by @rachelaronoff in https://github.com/rwb27/openflexure_microscope/issues/43#issuecomment-472307883_

rwb27, 2019-03-13:
Hi @rachelaronoff, you do need one extra lens to make it work properly; the design is meant to use a "finite conjugates" microscope objective, together with a secondary lens to shorten the distance to the sensor and adjust the image size to fit the Pi camera. The two pre-compiled options are:

    a 12.7mm diameter, 50mm focal length achromatic lens, e.g. ThorLabs AC127-050-A - fits f50d13
    a 16mm diameter, 40mm focal length lens, e.g. Comar 40 PC 16 (I think) - fits f40d16
    I reccommend the ThorLabs lens, it is slightly more expensive, but gives better quality images.

If you have an "infinity corrected" lens, you are lucky (these are generally more expensive and better quality) but it won't work properly with either of the options; you will still need one of the above lenses, but would need to compile a custom optics module. If that's the case, let me know and I'll compile one for you - it would be no bad thing to add this to the build script.

If you are not sure whether your objective is infinity corrected or finite conjugates, look at the writing on the side of it. If the number 160 appears anywhere, it is probably a standard finite-conjugates objective. If the infinity symbol is present, it's probably infinity corrected. Most RMS-threaded objectives (i.e. ones that will fit the plastic part) are finite conjugates. Modern infinity-corrected lenses tend to have larger threads.

Lastly, you can build the microscope without the additional lens. It will still produce images, but they will be of lower quality and the magnification will be different. I don't reccommend this, but it's possible it will work for your application.

closed 2019-03-13 by rwb27

MakerTobey, 2019-03-14:
Reference to follow up question:
https://github.com/rwb27/openflexure_microscope/issues/43#issuecomment-472930303 
And response:
https://github.com/rwb27/openflexure_microscope/issues/43#issuecomment-472931448

I guess this is where I start using OpenSCAD :-)