rwb27/openflexure_microscope #113: Manual to build SS and LS65 microscopes

nikitajes, 2019-03-14:
[Building the microscopes.docx](https://github.com/rwb27/openflexure_microscope/files/2966790/Building.the.microscopes.docx)

rwb27, 2019-03-14:
Thanks for uploading these - it's a very nice documentation of what you've built.  It looks like you were using a slightly older version of the microscope, so the parts (and instructions) have improved since then, and actually look a bit more like what you've put together, I think.

Do you have the original image files somewhere, and if so would you mind uploading a zip of them?  It is always useful to have good-quality images.  I think just exporting the word file as HTML would probably do it...

Thanks!

rwb27, 2019-04-30:
Thanks again for these.  

The repository is [moving to GitLab](https://gitlab.com/openflexure/openflexure-microscope/), and this issue has been migrated.  I'll close all the issues here in due course, but am leaving notices on all the currently-open ones.  If you head over to the other repository, this issue will be updated there.

closed 2019-04-30 by rwb27